meshtastic/design #9: Conditional Field Validation

garthvh, 2024-01-21:
Would it be good to have common field validation across platforms so that conflicting settings are not enabled by users? 

* Encryption and JSON mode in MQTT Can not be enabled at the same time
* Client proxy and JSON mode can not be enabled at the same time
* Both phone and Device GPS should not be enabled at the same time.
* Wifi, MQTT and Client Proxy should not be enabled at the same time
* Sections with client and sender enabled options (detection sensor, range test) should not expose settings not related to that device's role.

b8b8, 2024-01-21:
"Both phone and Device GPS should not be enabled at the same time."

Is there not a legitimate use case for this though. If you have higher quality phone GPS use it, but then default back to node GPS automatically. My car node (when i am actively driving around) is bluetoothed to my phone and providing high quality GPS updates to the device. When i am parked or someone else is using my car that doesnt have meshtastic, the device will use the onboard GPS and act as a "tracker" around town.

garthvh, 2024-01-21:
> Is there not a legitimate use case for this though. If you have higher quality phone GPS use it, but then default back to node GPS automatically. My car node (when i am actively driving around) is [...]

Yeah this one is more complicated for sure.  Smart position actually works really well, you can prevent blended positions pretty easily and fall back to device gps when the phone disconnects, but all the other modes just blend the data together.

b8b8, 2024-01-21:
Blending is good. Just dont want a situation where it sets to phone position, toggles off discrete GPS, then loses phone position and doesnt turn discrete GPS on again.

garthvh, 2026-11-08:
The single-field rules moved into the schema: protobufs `field_metadata.proto` carries `diy_only` and `oem_fixed_gps` (hide device GPS settings on boards with fixed GPS), `min_value`/`max_value`, `admin_only`, and `since_firmware`/`deprecated_since`, and Apple and Android both read it.

Rules between fields (proxy vs JSON, WiFi vs MQTT vs proxy, LoRa settings under a preset, role-specific module settings) can't be expressed there, since attributes are single values, so they stay in each client's config screens. If a specific conflict still causes trouble, please open a focused issue, or a protobufs proposal for a conditional attribute.

closed 2026-11-08 by garthvh