meshtastic/design #15: Signal Meter

garthvh, 2024-05-06:
In the Apple apps there is a signal meter Based on https://sensing-labs.com/f-a-q/a-good-radio-level/ and using the individual SNR values from here for the different spreading factors https://www.thethingsnetwork.org/docs/lorawan/rssi-and-snr/

ianmcorvidae, 2024-05-31:
Right now, RSSI isn't something that's stored in NodeDB, which makes this challenging to implement in the python CLI (until/unless I add some sort of persistence, but even that's sort of questionable since most uses of the CLI don't see that many received packets, so it'd often be very out-of-date info). I'm not sure if that's something that should be added to NodeDB or if our cross-platform focus should be mostly on the SNR side of this. Thoughts?

garthvh, 2024-09-02:
> Right now, RSSI isn't something that's stored in NodeDB, which makes this challenging to implement in the python CLI (until/unless I add some sort of persistence, but even that's sort of [...]

This is s good point, removing the CLI's as persistence is definitely required for this.

garthvh, 2026-09-03:
Closing: Apple and Android now implement the same signal-quality algorithm, including the noise-floor blend decided on 2026-08-12.

**The decision** (recorded on the Web child issue meshtastic/web#1241): when the receiving node has reported a noise floor, rate both `snr − presetLimit` and `(rssi − noiseFloor) − presetLimit` and take the more conservative tier; fall back to the preset-relative SNR + fixed-RSSI blend when no noise floor is available.

**Apple** — `getLoRaSignalStrength(snr:rssi:preset:noiseFloor:)` in `LoRaSignalStrengthIndicator.swift` computes both tiers and returns `min(snrTier, rssiTier)`, falling back to `legacyRssiSnrRating` (fixed -115/-120/-126 thresholds combined with SNR) when the noise floor is absent or zero.

**Android** — meshtastic/Meshtastic-Android#7009 (merged 2026-09-02, closing meshtastic/Meshtastic-Android#6826) adds a `LocalNoiseFloor` CompositionLocal sourced from the connected radio's `LocalStats`, normalizes firmware's `noise_floor == 0` sentinel to null, and takes the worse of the SNR margin and `(rssi − noiseFloor) − presetLimit`. Thanks @jamesarich.

**The LongSlow bug is also fixed** — meshtastic/Meshtastic-Apple#2041 (closed completed 2026-07-16). `snrLimit()` returned the SF7 floor (-7.5 dB) for LongSlow; it now returns -20 dB, matching Android's `LONG_SLOW` value. That was the last correctness gap between the two implementations.

**On the remaining checklist boxes:**
- **Web UI** — meshtastic/web#1241 stays open and tracks this independently. Cross-platform alignment is now scored on Apple + Android parity, so Web does not gate this issue.
- **Network Management Client** — dropped from scope in 2024. As @ianmcorvidae noted, RSSI isn't persisted in NodeDB and the CLI sees too few packets for a meaningful rating; that was accepted at the time.

Sub-issue #79 (signal strength from incoming packets) was already closed as satisfied on mobile.

closed 2026-09-03 by garthvh