meshtastic/design #21: Cross platform messaging features

thebentern, 2024-08-28:
- Reply to message
- "Tapback" emoji responses
- Github flavored markdown rendering
- Ability to @User with deep-link to node detail and notification options
   - Users can @ another node by typing the @ symbol Long name, short name, or user.ID
   - The actual text payload representation should package this as the User.Id: "@!ffccee11" for the wire
- PKI secured indicators for direct message threads to a node

tekstrand, 2024-08-29:
@thebentern is this supposed to be in `design`?

SpudGunMan, 2024-10-02:
There was discussions over on discord and documenting them under this issue as there is a lot of crossover. 

The discussion came about for a use case in having more interaction with the client. Specifically by enhancing the client applications for meshtastic with enhanced markup logic giving the user a clickable link which would perform a set of actions in the meshtastic client. 

The concept of a 🔕 is used to identify a non type-able character much like 🔔 is used today. 

The user experience would be something like a bot.py sending the text to a client `would you like to 🔕[draw](doDraw) or 🔕[play](doPlay)`

When a user clicked _draw_ or _play_ the meshtastic client would respond to the sender with `doDraw` or `doPlay` respectively.  bot.py would consume the Text 🃏

🦾
Another experience could be, an arduino library device which is attached to a node, I could send  `🔕send(runMotor(2))` from my iOS client to the mesh and the arduinoLibrayNode
Or serial module, would rx/consume/print `runMotor(2)` which can be processed by code for whatever. 

Some stuff the :no_bell: should do 
* present user with a hyperlink to click 
:no_bell:ClickMe
* useable in mobile client for manipulating robots and bots 

This would have a client action (ideas)
 * respond to sender with text from the link 
:no_bell:DrawCard (raw text is :no_bell:respond(draw)
* ask to set a new channel with PKI
:no_bell:newCh(PKI,name)
* ask to add admin PKI
:no_bell:newAdmin(PKI)
* provide any other ideas in the original GitHub issue 
* enable hidden text box user need to ID on phone to access or view contents (not enforced off mobile)

garthvh, 2026-11-08:
Everything under this parent is done or closed: editing (#45), translation (#83), full-text search (#75), tapback notifications (#35), and adjustable font size (#86, not planned because both apps follow the system text size). @mentions are on Apple and Android; Web's meshtastic/web#1242 doesn't gate alignment.

closed 2026-11-08 by garthvh