meshtastic/design #29: [Feature Request]: Per-node channel visibility tracking

ianmcorvidae, 2025-04-03:
### Contact Details

m_ia_n on discord

### Tell us your idea.

This would be potentially a large change, but which could provide some groundwork for some cool stuff in my opinion. What I'd like to see the app do is track what nodes it sees on what channels & with what sorts of settings. So for example, it might have the following information (formatted in an arbitrary way -- obviously this would be translated to database tables and such) for some node:

ID: !abcdef12
Channel: LongFast AQ==
 -> last seen: 2025-04-02T15:45:35-07:00
 -> last position precision: 16
Channel: Private <whatever key>
 -> last seen: 2025-04-03T11:00:53-07:00
 -> last position precision: 32

The goal here would be first of all to show the above information to the user, particularly when multiple channels are shared among several nodes (as is frequently the case if someone is using both the public channel and their own private channel). The down-the-road features would be things like more smartly choosing what channel to send things on. For example, in the above example a position request would probably in general be preferred on the private channel with the higher precision, but a traceroute would usually be better on the public channel that has more nodes to add their info to the trace. By tracking these properties for every node, the app is equipped to make these sorts of smarter channel-choice decisions. Or, if it is deemed UX-appropriate, this could allow the app to ask the user what channel to use, or allow the user the option of choosing even if it normally chooses on behalf of the user.

Firmware may need some updates to fully realize this, because at present you can't _force_ a message to go on the primary channel (you can with a secondary channel though). But tracking the information would be a useful step regardless, I think.

### Relevant log output

```shell

```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

thebentern, 2025-04-03:
@ianmcorvidae I think this is a good one for a cross-platform design discussion so I'm transferring to design.

ianmcorvidae, 2026-05-31:
A member of my local community has pointed out a more typical use case for this data, which would be showing a list of users in a channel/online (i.e. seen in the last longer vs shorter period, for us). I think such a thing would be built on the same kind of data I describe here.

garthvh, 2026-08-12:
Status (August 2026): no implementation on any client yet. Worth noting the shape of the problem before anyone starts: clients only observe the channel index of packets they receive, and index↔key mappings differ per node, so genuinely tracking "which nodes share which channel" needs either a firmware/protocol signal or heuristics with real false-positive risk. If there's still appetite, the next step is a protocol sketch rather than client UI.

ianmcorvidae, 2026-08-20:
Channel indexes are a local concept, so it doesn't matter if the index differs on a remote node. Clients can observe an instantaneous channel index and translate it to a more persistent identity (name & key, or hash-byte & key). Nothing in the protocol for any other feature keeps track of other nodes dropping in and out, so a feature like this one would probably just use the same "seen within a given timeframe" logic that's already used for things like noting when a node is online.

NomDeTom, 2026-08-20:
Posting this here while I remember: since position is particularly small in size (16-17 bytes of payload by my reckoning), for 3.0 we may want to consider encrypting the same information *for every channel a node possesses* and then sending all of these payloads in the same packet. Nodes would decrypt all of the payloads they could, and would take the highest precision value from them.

garthvh, 2026-11-08:
Closing: neither client tracks which channels each node is seen on, and nobody owns this. If it comes back, please start from a protocol sketch, since the 3.0 multi-channel idea above would change the design.

closed 2026-11-08 by garthvh