meshtastic/design #30: [Feature Request]: Assign battery chemistry to INA sensor to track alternate battery types

b8b8, 2025-05-15:
### Tell us your idea.

Instead of tracking alternative battery discharge curves in the firmware and the requisite complexity I propose adding it to the app.

Already under the power config section you can assign the INA_2xxx i2c address for measuring the battery voltage externally to the onboard battery gauge. I think we could easily add toggle/selector to choose a battery type that would then re-graph the battery percentage display in the app ONLY. No firmware changes are required.

The 3 most used battery types that i think should be added are: Li-ion, LiFePO4, LTO (sodium ion is also getting more popular). To keep things simple this would be just for parallel battery configurations, so the standard curves could be used. 

This info would be helpful to the many people that are using external charge controllers with different chemistries to still get a valid battery percentage on the device metrics log.

This info would/could be stored in the persistent node database the app now has. No info/changes are sent over the mesh or stored in the firmware. This also allows random users who take an educated guess on the chemistry of a mountain node in their town to be say LTO based, could update their app to match as well.

In the example below channel-2 is measuring the LTO battery voltage. You could hold down the channel-2 to have a popup give you the option to select the battery type for this channel.

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-05-15:
Needs to be looked at from a cross platform perspective.

b8b8, 2025-05-15:
> Needs to be looked at from a cross platform perspective.

Agreed, could get messy quickly if not.

garthvh, 2026-11-08:
Closing: no client has picked this up, and where the chemistry setting would be stored was never decided. Happy to reopen with a short spec if someone wants to own it.

closed 2026-11-08 by garthvh